mirko/SonOTA #76: Suggestion: relation sonoff-#### in script to mac address

arendst, 2017-12-18:
I just updated a Sonoff Dual R2 using SonOTA succesfully. Great tool!

I noticed in your script you relate to the new sonoff as ``sonoff-####``. This is the Tasmota default hostname made of the name ``sonoff`` and the last three hex digits of the mac address in decimal (how cryptic that may sound).

As I'm in the lucky situation that the firmware image contains my ssid and password it immediatly connects to my network using the new hostname which I then have to find.... 

Currently I'm using Fing on my mobile to find any device using the name sonoff-####. Another option is scanning the SonOTA debug log for the mac address found.

If not to much of a problem it might be useful to update the script to provide the correct hostname using the information from the debug log.

sillyfrog, 2017-12-19:
Thanks!

I'll check it out, should be easy to implement, I'll let you know how I get on.

sillyfrog, 2017-12-19:
I had a few mins, so checked it out just now - unfortunately the MAC address does not come through in any of the data it sends back (I was hoping it would be the `deviceid`). So the next strategy will be to try and pull the MAC address out when it's on the `FinalStage` SSID - but that'll take a little longer.