mirko/SonOTA #79: Final Stage : sending file "/ota/image_arduino.bin"

mjoetoe, 2017-12-28:
- Operating System: Mac OS 10.13.2
- Python Version:  3.6
- Sonoff model: /

[debug_1514500217.log](https://github.com/mirko/SonOTA/files/1592260/debug_1514500217.log)

keeps saying : The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all firewalls have been COMPLETELY disabled on your system.
..............................

firewall is disabled

korsmi, 2018-01-04:
i do have the same issue. firewall is disabled. i try it on windows 10. i will possibly setup a linux machine to see if that is than working. are there any solutions for this at the moment?

I try to flash a ps-15

frawau, 2018-01-08:
I have a similar issue. 

OS Linux Kubuntu 17.10
Python 3.6
Device Sonoff Touch US

First problem I had was that my laptop would not get an IP address, so I configured it to set the IP address "manually", and it seemed to work. I got the "FinalStage" SSID, and again I had to set the address manually to 192.168.4.2, and it looked like it worked, but the FinalStage SSID never dissappear.
It keeps on PUT and GET the arduino file.

[debug_1515416547.log](https://github.com/mirko/SonOTA/files/1611714/debug_1515416547.log)

sillyfrog, 2018-01-28:
This is something I have heard of happening before - unfortunately I have not seen it so not been able to investigate why it may happen. My guess is bad or slightly faulty flash, so it's not able to write the image to the flash reliably, so when it reboots, it's still on the FinalStage flash image.

frawau, 2018-01-29:
Thanks for the reply. I flashed it over serial. I'm good for now..