mirko/SonOTA #81: Bricked while sending file

vaclavek, 2017-12-30:
Hi, 
I have probably bricked the device. 

- Operating System: Windows 7 Ultimate, SP 1
- Python Version:  
- Sonoff model: Basic

[debug_1514591829.log](https://github.com/mirko/SonOTA/files/1594067/debug_1514591829.log)

Everything went OK, but suddenly appeared error message (look at attachned screenshot, it is not visible in log). Due to source code, it looks like some problem with IPs, but I don't have here anything special, so don't know.

[image]

I was able to connect to http://192.168.4.1 after that and set AP SSID and Password, but after reboot, Sonoff seems bricked. Probably image didn't upload OK? Is it repairable? I will have to reflash it with wires, right?

mjoetoe, 2017-12-30:
Same problem hopefully fixable:(

SQLSammy, 2018-01-01:
Similar problem, something during FinalStage gone wrong,
now after REeset i get sonoff-5015 WLAN, can connect and enter my WLAN credentials,
but then it does not appear

Is there a way to repair via WLAN or is it bricked?

hnyb4dg3r, 2018-07-21:
Theoretically you should still be able to flash the old fashioned way by using an FTDI USB to Serial converter and the TX and RX pins on the Sonoff board. DrZzs on YouTube has a few good examples.

sillyfrog, 2018-07-21:
@SQLSammy If you are getting the "sonoff-5015" SSID coming up, then (in theory) things have gone to plan. I suspect that bad flash maybe in play here (especially given the other one worked fine) as it's pretty unlikely for there to be a data error (due to the nature of TCP). If you upload your logs, I can take a look and see if there is anything obvious.