mirko/SonOTA #82: SonOTA.exe 32bit

reloxx13, 2017-12-30:
Heya @sillyfrog ,

i got feedback from a user that the exe is not working under 32bit.

I recreated it now with a python 32bit Version and this 32bit exe should work for both.

Would be nice if u add it to releases.

[sonota32.zip](https://github.com/mirko/SonOTA/files/1594394/sonota32.zip)

hansskip, 2018-02-08:
Hello reloxx13
During unzip I have got a  Norton virus notification 
[image]

May be you can explain the reason of the notification?

reloxx13, 2018-02-08:
Dude i dont know, the file does not has any certificates etc. and is made "anonymous".

if you dont trust the exe, install pyinstaller and create your own exe. im sure it will be shown as "unsecure" too. all files for creating it are in the sonota repo.

Some more infos in my PR:
https://github.com/mirko/SonOTA/pull/74

also, this is not a virus warning. as you can read this is a warning that this file is unknown by norton.

RudolfRendier, 2018-08-15:
Thanks for providing a 32-bit build. Worked fine on my Sonoff Dual R2.