mirko/SonOTA #83: SonOTA refuses to install with latest firmware 1.6.0

WJ4IoT, 2017-12-30:
- Operating System: Windos 10 Pro
- Python Version: Python 3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:04:45) [MSC v.1900 32 bit (Intel)] on win32
- Sonoff model: SonOff Basic
- Remark received 3 SonOff, 
  1st tested in eWelink, worked fine, installed SonOTA (thanks for making a patiened script)
  3rd device straight to SonOTA flawlessly install
  2nd tested in eWelink, and (my mistake updated to latest firmware 1.6.0), this device not able to install SonOTA see log. 

Observation: with the new eWelink firmware a long button press goes first in a different modus (no continious blink and no AP-modes) a second long button press goes in continious blink and AP-modes. The sonOTA script continious however not complete. 

Not sure if the blinking part is the difference because I am out of clean devices. :)  

```
Using the following configuration:
        Server IP Address: 192.168.0.107
        WiFi SSID: SomeWhereNearBy
        WiFi Password: ******************
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
.......................Current IPs: ['192.168.0.105', '10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10001f90f1",
    "apikey": "efda78c7-1b1b-4019-a7d9-a9d3ac44bb28",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "SomeWhereNearBy",
    "password": "******************",
    "serverName": "192.168.0.107",
    "port": 8443
}
Traceback (most recent call last):
  File "sonota.py", line 734, in <module>
    main()
  File "sonota.py", line 729, in main
    stage1()
  File "sonota.py", line 612, in stage1
    "http://10.10.7.1/ap", "POST", json.dumps(data))
  File "C:\Users\WJ\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1322, in request
    (response, content) = self._request(conn, authority, uri, request_uri, method, body, headers, redirections, cachekey)
  File "C:\Users\WJ\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1072, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "C:\Users\WJ\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1025, in _conn_request
    response = conn.getresponse()
  File "C:\Program Files (x86)\Python36-32\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Program Files (x86)\Python36-32\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Program Files (x86)\Python36-32\lib\http\client.py", line 258, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Program Files (x86)\Python36-32\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out
```
 Standard log (with less information):
[debug_1514638777.log](https://github.com/mirko/SonOTA/files/1594395/debug_1514638777.log)

reloxx13, 2017-12-31:
https://github.com/mirko/SonOTA/wiki#known-working-configurations

WJ4IoT, 2017-12-31:
a.k.a. RTFM, at least my expectations are correct.
Should  i close this comment because it is already captured in the specs or remain open because it is an unsolved issue?  Thanks for your patience and good work.

reloxx13, 2018-01-01:
i think this is not fixable even not in the future because we dont know the private key from ITEAD.
their is no way around :(

flash by usb flasher still works.

WJ4IoT, 2018-01-02:
Will break the seal and develop my solderskills, needless to say I love this OTA version!

closed 2018-01-02 by WJ4IoT