mirko/SonOTA #87: sonota

ukpaulthm, 2018-01-02:
Hi all
 Am running Windows and This is my first time I have ever had a go at flashing anything so please bear with me
 Am having problems been up half the night as I don’t like things beating me
 Am trying to flash sonoff basic with sonota I get too the point where the script asks me to hold the button in for 7 seconds untill green light is flashing then it asks me to connect via WiFi and as soon as I type in the password 12345678 the sonota command box closes on its own been trying for about 8 hours now and totally stuck why it’s doing this . Can anyone please help me
 I have managed to flash one out off 3 sonoff basics just don’t understand why the script is closing it’s screen on the other two sonoff
 Thanks
 Paul
2018-01-02 14:02:16,927: DEBUG: Current IPs: ['169.254.76.200']
2018-01-02 14:02:42,474: INFO: Using the following configuration:
2018-01-02 14:02:42,474: INFO: 	Server IP Address: 169.254.76.200
2018-01-02 14:02:42,474: INFO: 	WiFi SSID: My Internet
2018-01-02 14:02:42,474: INFO: 	WiFi Password: **************
2018-01-02 14:02:42,474: INFO: Platform: win32
2018-01-02 14:02:42,584: INFO: ** Now connect via WiFi to your Sonoff device.
2018-01-02 14:02:42,584: INFO: ** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
2018-01-02 14:02:42,584: INFO: To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
2018-01-02 14:02:42,584: INFO: ** This application should be kept running and will wait until connected to the Sonoff...
2018-01-02 14:03:03,892: DEBUG: Current IPs: ['192.168.0.25']
2018-01-02 14:03:16,637: DEBUG: Current IPs: ['10.10.7.2']
2018-01-02 14:03:16,715: DEBUG: ~~ Connection attempt
2018-01-02 14:03:16,715: DEBUG: >> HTTP GET /10.10.7.1/device
2018-01-02 14:03:16,731: DEBUG: << {
2018-01-02 14:03:16,731: DEBUG:     "apikey": "588e5e1d-7ba6-4af1-94c3-463bfac240fa",
2018-01-02 14:03:16,731: DEBUG:     "deviceid": "10001b106c",
2018-01-02 14:03:16,731: DEBUG:     "accept": "post"
2018-01-02 14:03:16,731: DEBUG: }
2018-01-02 14:03:16,731: DEBUG: >> HTTP POST /10.10.7.1/ap
2018-01-02 14:03:16,731: DEBUG: >> {
2018-01-02 14:03:16,731: DEBUG:     "port": 8443,
2018-01-02 14:03:16,731: DEBUG:     "version": 4,
2018-01-02 14:03:16,731: DEBUG:     "ssid": "My Internet",
2018-01-02 14:03:16,731: DEBUG:     "serverName": "169.254.76.200",
2018-01-02 14:03:16,731: DEBUG:     "password": "**************"
2018-01-02 14:03:16,731: DEBUG: }

ukpaulthm, 2018-01-02:
Just too add  the one that I flashed ok was running firmware version 1.5.5. I . The two that the command screen closed on me is on 1.6.0

JacoFourie, 2018-01-02:
1.6 does not work.

ukpaulthm, 2018-01-02:
DO you know if I can downgrade them ?

JacoFourie, 2018-01-02:
https://github.com/mirko/SonOTA/issues/58

JacoFourie, 2018-01-02:
https://www.itead.cc/sonoff-wifi-wireless-switch.html#comment-3672182614

closed 2018-01-28 by sillyfrog