mirko/SonOTA #88: [Errno 98] Address already in use

fabsss, 2018-01-04:
- Operating System: Linux/Windows
- Python Version: 3.5.3 (Linux), 3.5.4 (Windows)
- Sonoff model: S20

After connecting to my router the 2nd time (I successfully connected to the Sonoff-Device before as seen in the traces), the application crashes printing the following information:

~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Traceback (most recent call last):
  File "./sonota.py", line 734, in <module>
    main()
  File "./sonota.py", line 730, in main
    stage2()
  File "./sonota.py", line 653, in stage2
    app.listen(DEFAULT_PORT_HTTP)
  File "/home/pi/.local/lib/python3.5/site-packages/tornado/web.py", line 1943, in listen
    server.listen(port, address)
  File "/home/pi/.local/lib/python3.5/site-packages/tornado/tcpserver.py", line 142, in listen
    sockets = bind_sockets(port, address=address)
  File "/home/pi/.local/lib/python3.5/site-packages/tornado/netutil.py", line 197, in bind_sockets
    sock.bind(sockaddr)
OSError: [Errno 98] Address already in use

The debug*.log shows the following information:

2018-01-04 19:28:30,499: DEBUG: Current IPs: ['192.168.178.49']
2018-01-04 19:28:43,181: INFO: Using the following configuration:
2018-01-04 19:28:43,181: INFO: 	Server IP Address: 192.168.178.49
2018-01-04 19:28:43,182: INFO: 	WiFi SSID: Martin Router King
2018-01-04 19:28:43,182: INFO: 	WiFi Password: *****************
2018-01-04 19:28:43,183: INFO: Platform: linux
2018-01-04 19:28:43,185: INFO: ** Now connect via WiFi to your Sonoff device.
2018-01-04 19:28:43,185: INFO: ** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
2018-01-04 19:28:43,186: INFO: To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
2018-01-04 19:28:43,186: INFO: ** This application should be kept running and will wait until connected to the Sonoff...
2018-01-04 19:29:03,225: DEBUG: Current IPs: []
2018-01-04 19:29:11,241: DEBUG: Current IPs: ['10.10.7.2']
2018-01-04 19:29:18,453: DEBUG: ~~ Connection attempt
2018-01-04 19:29:18,454: DEBUG: >> HTTP GET /10.10.7.1/device
2018-01-04 19:29:18,482: DEBUG: << {
2018-01-04 19:29:18,483: DEBUG:     "deviceid": "100017eb8a",
2018-01-04 19:29:18,485: DEBUG:     "accept": "post",
2018-01-04 19:29:18,486: DEBUG:     "apikey": "eace0107-ad53-4358-9782-f561e7b69abd"
2018-01-04 19:29:18,486: DEBUG: }
2018-01-04 19:29:18,488: DEBUG: >> HTTP POST /10.10.7.1/ap
2018-01-04 19:29:18,489: DEBUG: >> {
2018-01-04 19:29:18,490: DEBUG:     "password": "*****************",
2018-01-04 19:29:18,490: DEBUG:     "version": 4,
2018-01-04 19:29:18,491: DEBUG:     "port": 8443,
2018-01-04 19:29:18,491: DEBUG:     "serverName": "192.168.178.49",
2018-01-04 19:29:18,492: DEBUG:     "ssid": "Martin Router King"
2018-01-04 19:29:18,492: DEBUG: }
2018-01-04 19:29:18,660: DEBUG: << {
2018-01-04 19:29:18,661: DEBUG:     "error": 0
2018-01-04 19:29:18,661: DEBUG: }
2018-01-04 19:29:18,661: INFO: ~~ Provisioning completed
2018-01-04 19:29:18,663: INFO: Starting stage2...
2018-01-04 19:29:18,667: INFO: ** The IP address of <serve_host> (192.168.178.49) is not assigned to any interface on this machine.
2018-01-04 19:29:18,668: INFO: ** Please change WiFi network to Martin Router King and make sure 192.168.178.49 is being assigned to your WiFi interface.
2018-01-04 19:29:18,668: INFO: ** This application should be kept running and will wait until connected to the WiFi...
2018-01-04 19:29:20,672: DEBUG: Current IPs: []
2018-01-04 19:31:18,854: DEBUG: Current IPs: ['192.168.178.49']
2018-01-04 19:31:18,855: INFO: ~~ Starting web server (HTTP port: 8080, HTTPS port 8443)

82vmayer, 2018-01-06:
Operating System: MAC
    Python Version: 3.6
    Sonoff model: Sonoff 4CH

Hello,

i have the same issue.  The script was able to connect to Sonoff-Device and crashed after my mac reconnected  2nd time the application crash during starting the web server (HTTP port: 8080, HTTPS port 8443). Shell output:

Using the following configuration:
	Server IP Address: 192.168.178.105
	WiFi SSID: va-ttm
	WiFi Password: ********************************
Platform: darwin
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
....Current IPs: []
..............Current IPs: ['192.168.178.105']
.......................Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "1000103d6b",
    "apikey": "a0814e35-28cf-4927-b800-e47b3c52270f",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "va-ttm",
    "password": "********************************",
    "serverName": "192.168.178.105",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.178.105) is not assigned to any interface on this machine.
** Please change WiFi network to va-ttm and make sure 192.168.178.105 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
......Current IPs: []
..Current IPs: ['192.168.178.105']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Traceback (most recent call last):
  File "sonota.py", line 734, in <module>
    main()
  File "sonota.py", line 730, in main
    stage2()
  File "sonota.py", line 653, in stage2
    app.listen(DEFAULT_PORT_HTTP)
  File "/Users/valentin/Library/Python/3.6/lib/python/site-packages/tornado/web.py", line 1943, in listen
    server.listen(port, address)
  File "/Users/valentin/Library/Python/3.6/lib/python/site-packages/tornado/tcpserver.py", line 142, in listen
    sockets = bind_sockets(port, address=address)
  File "/Users/valentin/Library/Python/3.6/lib/python/site-packages/tornado/netutil.py", line 197, in bind_sockets
    sock.bind(sockaddr)
OSError: [Errno 48] Address already in use

sillyfrog, 2018-01-28:
This is another service or program running on the ports (8080 most likely, or maybe 8443). If you exit all other services and programs, it will likely start working. If you still have the issue, look on the web for ways to see what programs are using what ports, and find what it is using 8080 and 8443, then kill it, and run this again.

Good luck!

jnsdrtlf, 2018-08-07:
@sillyfrog this fixed the issue. You can close this now.

closed 2018-08-08 by sillyfrog

robertnaquila, 2018-09-15:
Having the same issue. Below are the programs using 8080.

`COMMAND PID   USER   FD   TYPE             DEVICE SIZE/OFF NODE NAME
nginx    52   root    6u  IPv4 0xff4faa4803cc14e3      0t0  TCP *:http-alt (LISTEN)
nginx   148 nobody    6u  IPv4 0xff4faa4803cc14e3      0t0  TCP *:http-alt (LISTEN)`

Tried to kill them but got an error.

`Roberts-MacBook-Pro:~ robert.naquila$ kill -9 52
-bash: kill: (52) - Operation not permitted
Roberts-MacBook-Pro:~ robert.naquila$ kill -9 148
-bash: kill: (148) - Operation not permitted`

Help is much appreciated.

jnsdrtlf, 2018-09-15:
use `sudo kill -9 <pid>` or stop the `nginx` service with `sudo launchctl unload /Library/LaunchDaemons/homebrew.mxcl.nginx.plist` (assuming you installed `nginx` using `brew`)

robertnaquila, 2018-09-15:
That worked! Thank you so much!