mirko/SonOTA #91: Stuck ;(

tukestolo, 2018-01-04:
- Operating System:  max os
- Python Version: python 3
- Sonoff model: Rf bridge

Hello friends, I am stuck in this fase I don't know whats happen. 
I will describe the steps I have done.

1 - I have run sonota.py
2 - I put the only Ip that appear on console, them I put my wifi ssid and password of my hotspot that I created on my phone.
3 - I put the rf on ap mode and connect to it ...
4 - he do the things and automatic disconnect from the wifi.
5 - i connect again to my hotspot
6 - Always stuck on this step....

The Sonoff don't connect to my hotspot and I don't know why :/

Please help me ...

2018-01-04 23:30:34,982: INFO: *** IMPORTANT! ***
2018-01-04 23:30:34,982: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2018-01-04 23:30:34,983: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2018-01-04 23:30:34,983: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2018-01-04 23:30:34,983: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2018-01-04 23:31:35,366: INFO: *** IMPORTANT! ***
2018-01-04 23:31:35,366: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2018-01-04 23:31:35,366: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2018-01-04 23:31:35,366: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2018-01-04 23:31:35,366: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2018-01-04 23:32:35,762: INFO: *** IMPORTANT! ***
2018-01-04 23:32:35,762: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2018-01-04 23:32:35,762: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2018-01-04 23:32:35,762: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2018-01-04 23:32:35,762: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.

sillyfrog, 2018-01-28:
Can you please attach the full log file - it will be saved into the sonota folder after each run, it's earlier in the process that may help me.

Also be aware of the issue if it's running firmware v1.6, they have implemented certificate pinning so this process will no longer work if you have update :(   See #58 for more details.

bachya, 2018-04-03:
@sillyfrog Hopping around a few different issues, but would love your thoughts. I'm in a similar position: attempted to use SonOTA, got stuck at "Final Stage", and figured I had to go down the serial route (detailed [here](https://github.com/arendst/Sonoff-Tasmota/issues/2344)). That said, I'm having a heck of a time getting the manual flashing to work – does anything special have to be done, knowing that "Final Stage" occurred and _something_ was transferred to the device?

sillyfrog, 2018-05-07:
Do you know what firmware version you are using? See the wiki as with the latest firmware from ITEAD it's not possible to using SonOTA any more :(