mirko/SonOTA #95: Sonoff Dual R2

donoo, 2018-01-07:
- Operating System: Windows 10 
- Python Version:  3.5.4
- Sonoff model: Sonoff Dual R2

Hi,

I have flashed Sonoff Dual R2 using SonOTA. Everything is working fine. But I m not able to find options for enabling physical button as documented here : https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual-and-Dual-R2.

 I can only see following options in Config Page:

[image]

Is there way to enable these options while using SonOTA

[image]

Thanks in advance!

Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.

AlessandroEmm, 2018-01-09:
Have you updated to the latest release of tasmota? i think you should be asking there...

donoo, 2018-01-10:
Hi,
Yes, after updating to latest version 5.11.1 i can see all options.
Thanks for the reply!

closed 2018-01-10 by donoo