mirko/SonOTA #97: Update Original Firmware 1.5.5 to 1.6.0

thiagoarreguy, 2018-01-08:
- Operating System: Mac OS X
- Python Version: 3.6.4
- Sonoff model: basic one channel

I had update the original firmware from 1.5.5 to 1.6.0 and now im having this problem with sonota.py

###########################################################################
Current IPs: ['192.168.1.123']
Select IP address of the WiFi interface:
    0: 192.168.1.123
Select IP address [0]: 0
WiFi SSID: mywifi
WiFi Password: mypassword

Using the following configuration:
	Server IP Address: 192.168.1.123
	WiFi SSID: 660-101-24GHZ
	WiFi Password: ************
Platform: darwin
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
.......................................................Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100015516d",
    "accept": "post",
    "apikey": "5e5b60dd-a76a-4f04-8a7e-15d37128f3b2"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "serverName": "192.168.1.123",
    "ssid": "660-101-24GHZ",
    "version": 4,
    "port": 8443,
    "password": "************"
}
Traceback (most recent call last):
  File "/Users/thiagoarreguy/Desktop/mqtt_sonoff/SonOTA-master/sonota.py", line 734, in <module>
    main()
  File "/Users/thiagoarreguy/Desktop/mqtt_sonoff/SonOTA-master/sonota.py", line 729, in main
    stage1()
  File "/Users/thiagoarreguy/Desktop/mqtt_sonoff/SonOTA-master/sonota.py", line 612, in stage1
    "http://10.10.7.1/ap", "POST", json.dumps(data))
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/httplib2/__init__.py", line 1322, in request
    (response, content) = self._request(conn, authority, uri, request_uri, method, body, headers, redirections, cachekey)
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/httplib2/__init__.py", line 1072, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/httplib2/__init__.py", line 1025, in _conn_request
    response = conn.getresponse()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/http/client.py", line 1171, in getresponse
    response.begin()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/http/client.py", line 351, in begin
    version, status, reason = self._read_status()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/http/client.py", line 313, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/socket.py", line 374, in readinto
    return self._sock.recv_into(b)
###########################################################################

reloxx13, 2018-01-08:
read the wiki

#close

larrybell, 2018-01-12:
any idea if the will get fixed? Is there a way to rollback the firmware on the Sonoff?

thiagoarreguy, 2018-01-12:
you can try writeback the original firmare with esptool if you have the
original firmware.bin

reloxx13, 2018-01-12:
Note: as far as i know the dumped firmware is unique, means u cannot use another orginal dumped firmware from another device.

sillyfrog, 2018-01-28:
I have not tried it, but as @reloxx13 mentioned, you'll need to flash via serial, so going straight to the final image is a better way to go. (PS: if you have not already, please go to http://disq.us/p/1oqbm8m and post a request to ITEAD - one day they may listen!)

closed 2018-01-28 by sillyfrog