mirko/SonOTA #98: No progress after connecting to FinalStage Wifi

LuckyTGX, 2018-01-11:
- Operating System: Windows 7 (32Bit)
- Python Version: 3.6
- Sonoff model: Sonoff Touch

I got stuck once at the FinalStage and tried it now several times to resume. I always get this:
C:\Sonoff\SonOTA-master>python sonota.py
Current IPs: ['10.0.0.124', '192.168.4.2']
Select IP address of the WiFi interface:
    0: 10.0.0.124
    1: 192.168.4.2
Select IP address [0]: 1
It looks like you are already on a Sonoff/Final stage WiFi network, please chang
e to your normal WiFi network. If this has already been done, you may need to mo
dify the IP range of your LAN to use this tool safely. If this is what you inten
ded run again with --no-prov to skip this step.
Quitting.

C:\Sonoff\SonOTA-master>python sonota.py --no-prov
Platform: win32
Starting stage2...
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect
Current IPs: ['10.0.0.124', '192.168.4.2']

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
.........................

So I get the FinalStage-Wifi when I plug in my Sonoff-Touch. I can connect to the FinalStage-Wifi with my computer, but still it seems that the Python-Script does not recognize that I am actually connected - or it just does not continue for any other reason.

Thanks in advance for helping me out! I have absolutely no clue what to do now.

Best regards,
Lucky

LuckyTGX, 2018-01-11:
I need to add something that surprisingly just happened for the first time:

..........................Current IPs: ['10.0.0.124', '169.254.26.61']

It appears we have been disconnected from the "FinalStage" SSID, however the fin
al image has not been downloaded. Reconnect to "FinalStage" when it returns to c
ontinue the process (this may require a power cycle of your Sonoff device)...

................................................................................
..........................................

Curiously I did not need to restart the sonoff or reconnect the wifi. At least I did not notice that the wifi has been disconnected. When I checked, it was still connected.

It is still running now and adding dots to the command prompt. I have no clue how long this will continue.

Can anybody give me an average amount of minutes it usually takes? I am waiting >5 mins now for sure.

Best regards,
Lucky

LuckyTGX, 2018-01-11:
Since it really disconnected finally .. I restarted the sonoff (power off and on) and reconnected the wifi to FinalStage.

Now this continues:
................................................................................
................................................................................
........................................................................Current
IPs: ['10.0.0.124', '192.168.4.2']
....

The "FinalStage" SSID will disappear when the device has been fully flashed and
image_arduino.bin has been installed.
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all fire
walls have been COMPLETELY disabled on your system.
.....

It does not say "Sending file...", so I guess I am really stuck here.

My windows firewall and virus-scan etc. is all off.

Best regards,
Robert

erwill2, 2018-01-13:
try legacy option

mjoetoe, 2018-01-14:
Same problem , someone pls help

Herschdorfer, 2018-02-16:
similar to #107, #103

HFsi, 2019-02-09:
Same on Ubuntu