mirko/SonOTA #103: Two POW stuck at final stage

thefathefa, 2018-01-27:
- Operating System: Linux
- Sonoff model: POW

Hi, I have already flashed several POW and Basic but for 2 of the POWs I got stuck in the following state :

I heard the beeps on the device

I saw several GETs of /ota/image etc... But I did not keep traces :-(

LEDs are now always off (even after multiple presses and long ones - but there is a super short led on when I power cycle it). And pressing the button doesn't toggle the test lights.

FinalStage WiFi is on and I can connect to it, but SonOTA stays stuck at "If there is no "Sending file: /ota/image_arduino.bon" log entry etc..."

I power cycled the devices

I have tried restarting SonOTA with --no-prov option, I can confirm that the PC gets given 192.168.4.2 as IP address, but when I try to connect to 192.168.4.1, I get ERR_CONNECTION_REFUSED

I know the firewalls are ok because I was able to flash a device (after those 2 got stuck, so my config is now probably different from what it was when the flash did not work on those 2 devices, but I know it is now ok).

Is there hope for me to recover them without having to go to a wired connection?
What would you suggest I look at, please?

Thanks in advance for your help and thanks a lot for the great job on SonOTA!

sillyfrog, 2018-01-28:
When it's on FinalStage, the Sonoff is connecting back to your PC to download the image, so that's expected. It should be retrying every few minutes.

I have heard of this happening before, where it does not attempt to download - unfortunately I'm not sure what causes it - the most likely suspect is bad (or slightly faulty) flash, as during this process it is rewriting the flash memory.

thefathefa, 2018-01-28:
ok, thanks for the feedback. So I'll ned to go to the wires way:-)

magnusoverli, 2018-02-07:
Did you sort it out using cables? I am also at this issue, but I don't have cables to fix it...

thefathefa, 2018-02-07:
yes, I did.
You'll find on internet many tutorials on how to do this and to buy (few euros).
One thing : there is no need to bother with soldering. You can hold the things in the right position while the transfer takes place.

Herschdorfer, 2018-02-16:
might that be the same as my #107 ?

I have now two T1 with the exact same behavior.

tholterhus, 2018-03-11:
@thefathefa could you pls post a link to the tutorial you used? I have wires and I can connect to the console ... I'm on windows, do I need to be on linux?

thefathefa, 2018-03-11:
Look at this one : https://www.youtube.com/watch?v=pVPPiYAo8NI 
The first part is I think what you are looking for.

tali1974, 2018-04-09:
upgrade the POW to 2.0.4. and it worked for me.