mirko/SonOTA #105: Stop asking ITEAD to "fix" it

neuhaus, 2018-01-31:
I don't think it's a good idea to ask your users to complain to ITEAD to "fix" their device in the README.

It's not a fix. It's re-introducing a security hole.

There is a workaround. It requires soldering.

We should encourage better security, not ask for weaker security!

alanvanwyk, 2018-01-31:
Disagree, 

They could simply enable a switch on their devices to allow override of the cert issue – many providers have this as ‘debug mode’ or ‘dev mode’ – even something like an Amazon Firestick lets you decide!

I don’t fell that being forced to use a 3rd party app via an intermediary to manage my devices on an unknown cloud is more secure than being able to install a locally managed mqtt engine.
I say keep badgering them, until they allow us to decide for ourselves!

From: Sven Neuhaus
Sent: 31 January 2018 15:13
To: mirko/SonOTA
Cc: Subscribed
Subject: [mirko/SonOTA] Stop asking ITEAD to "fix" it (#105)

I don't think it's a good idea to ask your users to complain to ITEAD to "fix" their device in the README.
It's not a fix. It's re-introducing a security hole.
There is a workaround. It requires soldering.
We should encourage better security, not ask for weaker security!

TekMason, 2018-02-01:
bulletprooffool is right on with his comment.
Improving **default** security is almost always a good thing but offering an option for makers to hack Itead devices is a necessity.  Itead's success can be directly linked to embracing the maker community and allowing (even encouraged) firmware updates.
Another case study in enabling makers  [The WRT54GL: A 54Mbps router from 2005 still makes millions for Linksys](https://arstechnica.com/information-technology/2016/07/the-wrt54gl-a-54mbps-router-from-2005-still-makes-millions-for-linksys/
) 
TekMason

sillyfrog, 2018-02-01:
Fair point - and I was probably a bit frustrated when putting that on the home page, I have now updated. For an item that designed for DIY, allowing users to downgrade is critical I think. The "downgrade firmware" option I think will be the ideal balance, people can downgrade if they know what they are doing, but it's secure by default, and it requires no changes in hardware.

closed 2018-02-01 by sillyfrog

LucaAghemo, 2018-02-18:
I agree with sillyfrog, downgrade might be a fair solution.
To me, btw, they should sell a version already eanbled for mqtt use. That might be the best way.
Another good solution might be to give API so that home automation system can create a Ewelink module.
There is a post abut that in Home Assistant community, and someone by Itead has even answered they might be available. Not the best, because it still require an internet connection and relaying to Ewelink cloud, but better than soldering IMHO. Also, this solution wouldn't void warranty.