mirko/SonOTA #107: T1 Stuck in FinalStage

Herschdorfer, 2018-02-15:
- Operating System: Debian
- Python Version: 3.4.2
- Sonoff model: T1

My T1 was flashing without issues but got stuck at the FinalStage upload and i (this was maybe too early) reset the device... :/

The FinalStage WLAN keeps appearing but i have no idea how to deliver the firmware. Is there any option to start sonota from the final stage phase?

[debug_1518730095.log](https://github.com/mirko/SonOTA/files/1729610/debug_1518730095.log)

Herschdorfer, 2018-02-16:
I found out that the --mo-prov switch is what i want, but the T1 is not downloading the image.

I can see the final stage SSID and can connect too. I'm getting the right IP address but the device is not accessing the image

neuman1812, 2018-02-16:
Is it possible your sonoff has the 1.6 firmware installed?  That was the same issue I had with mine.  It stops at the final stage.

Herschdorfer, 2018-02-16:
no, the both T1 came with 2.0.1. According to the Wiki they work. (At least the 2 channel)

For me the FinalStage SSID appeared very fast but does not go further.
The log actually said the device tired to download the file. The first device log does not show a HTTP 200 but my second device did show this.

Herschdorfer, 2018-02-18:
I just connected a T1 with the UART and get the folloing at boot:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : DOUT
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user2 @ 81000
```

Then i need to switch to 115200baud and get:

```
   Initalizing..................................................................                           .....................................Done
Current rom: 2
              Rom 1 magic byte: 0xE9
                                    Reflashing rom: 0
                                                     Configuring Wifi FinalStage...
                                                                                   Waiting for connection...
.Connected
Web server accessible...
Flash Mode: 3
ROM or Flash Mode not recognized
```

tholterhus, 2018-03-11:
I'm stuck at the exact same message on my S20.

Now what? How can I reflash this stage?

Herschdorfer, 2018-03-12:
I was not able to solve this issue and flashed via cable.

bachya, 2018-04-03:
I'm in the same position, but can't seem to flash the Sonoff manually after this "FinalStage block" (detailed [here](https://github.com/arendst/Sonoff-Tasmota/issues/2344)) – did you have to do anything special to flash via serial after having tried SonOTA?

Herschdorfer, 2018-04-09:
no, just connect the cables and flash, nothing needed otherwise.

(disconnect the main board from the relay board before flashing).

closed 2018-04-09 by Herschdorfer

gshinman, 2018-08-15:
I am trying to flash the T1 after stuck in Final Stage mode. I am unable to get into Flash mode although I have tried all the switch combinations documented. When connecting the 3.3v no lights go on on the T1 and I notice the voltage on the Vcc/Gnd drops to ~1v when connected - is this normal? Any suggestions? My first time flashing Tasmota to a Sonoff not via OTA
Thanks