mirko/SonOTA #108: Kesen 2 gang switch went... almost fine! Need some help make it perfect.

LucaAghemo, 2018-02-17:
I got this 2 gang Sonoff clone from a friend of mine:
https://www.amazon.it/gp/product/B076CJHPQ6/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1
I decided to open it, and see if I could flash it but... no luck. It has no reset button, the pin are forwarded on the PCB but... under the touch button 1! I tryed a last, desperate thing, flashing it with SonOTA, under OSX 10.9.5 and Python 3.6.4. 
Guess what: it worked! Almost...
Everything seems to work ok, mqtt connection, switching, I even connected it to Home Assistant. The only weird thing is that the touch button 2 doesn't work anymore. The relais 1 and 2 can be controlled via software or via the webserver page, everything connect and exchange informations, if I switch in the webserver page mqtt read and Home Assistant update, button 1 work perfectly, button 2... dead. This happen weather I power only the logic board with 3.3v or the whole thing assembled with 220v.
Quite a mistery to me, it used to work when standard, I tried 1 min before flashing, can't understand why. 
I attach the debug log, but it seems to me that everything went well in flashing.
Anyone can help? It would be nice to add another device to the compatibility list! :-)
Thanks in advance.
[debug_1518875169.log](https://github.com/mirko/SonOTA/files/1733815/debug_1518875169.log)

sillyfrog, 2018-02-22:
It sounds like you got Sonoff-Tasmota on there OK, so this maybe a question for that project. I would suggest trying different "Module Types" and see if they work any better - it maybe a minor tweak to get this bit going in Sonoff-Tasmota. Now that it's installed, you can just use the web interface to upload new versions as well (although be careful, as it sounds like it maybe a one way ticket if it goes bad). Good luck!

closed 2018-02-22 by sillyfrog

LucaAghemo, 2018-02-23:
Thank you sillyfrog, I already tried almost any module type, but the only thing I obtain is that, sometimes, the first button stops working too! 
I posted the same on Tasmota's Issues section, I think it needs some tweak on the code and reflashing via web. We'll see! :-)