mirko/SonOTA #113: T1 EU - Process finished with exit code 1

soytoledano, 2018-03-03:
- Operating System: Windows 7
- Python Version: 3.7.0a2
- Sonoff model: T1 EU Firmware version 1.7.0

[debug_1520103909.log](https://github.com/mirko/SonOTA/files/1777795/debug_1520103909.log)

Traceback (most recent call last):
  File "C:/Users/laura/PycharmProjects/sowpow/sonota.py", line 678, in <module>
    main()
  File "C:/Users/laura/PycharmProjects/sowpow/sonota.py", line 673, in main
    stage1()
  File "C:/Users/laura/PycharmProjects/sowpow/sonota.py", line 560, in stage1
    "http://10.10.7.1/ap", "POST", json.dumps(data))
  File "C:\Users\laura\AppData\Local\Programs\Python\Python37-32\lib\site-packages\httplib2\__init__.py", line 1322, in request
    (response, content) = self._request(conn, authority, uri, request_uri, method, body, headers, redirections, cachekey)
  File "C:\Users\laura\AppData\Local\Programs\Python\Python37-32\lib\site-packages\httplib2\__init__.py", line 1072, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "C:\Users\laura\AppData\Local\Programs\Python\Python37-32\lib\site-packages\httplib2\__init__.py", line 1025, in _conn_request
    response = conn.getresponse()
  File "C:\Users\laura\AppData\Local\Programs\Python\Python37-32\lib\http\client.py", line 1322, in getresponse
    response.begin()
  File "C:\Users\laura\AppData\Local\Programs\Python\Python37-32\lib\http\client.py", line 296, in begin
    version, status, reason = self._read_status()
  File "C:\Users\laura\AppData\Local\Programs\Python\Python37-32\lib\http\client.py", line 257, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\laura\AppData\Local\Programs\Python\Python37-32\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
ConnectionAbortedError: [WinError 10053] An established connection was aborted by the software in your host machine

Process finished with exit code 1

soytoledano, 2018-03-03:
After several resets I tired again and now FinalStage SSID doesn´t show up:

It seems stage 1 took only a few ms...

[debug_1520104833.log](https://github.com/mirko/SonOTA/files/1777873/debug_1520104833.log)

sillyfrog, 2018-03-03:
That first error is weird, however the second looks like #58 . I would encourage you to contact ITEAD (see the home page) and request they allow downgrades of the firmware - the more requests the better! :)

closed 2018-03-03 by sillyfrog