mirko/SonOTA #117: No Dual R2 option in Module confiq

ThunderMerweS, 2018-04-01:
- Operating System: Win 10
- Python Version: 3.65
- Sonoff model: Dual R2
Hello all,

Bit new to flashing firmware, but here goes.

So after doing the firmware flashing, all was good. However, when it comes to selecting the module, I do not find the Dual R2 Module, just the normal Dual

If I choose the Dual module, I only have 2 GPIO options, and if I change them both to be switches the toggle off/on doesn't do anything.

[image]

Rapejim, 2018-05-12:
This issue isn't of SonOTA.

Update to lastest version of Tasmota (https://github.com/arendst/Sonoff-Tasmota) and try again.

Olesatgt, 2018-05-18:
Super idea!

cpfarher, 2018-08-23:
Which are the steps for update? In the documentation said: "If a new release of Sonoff-Tasmota is available and you wish to use it, download the binary from the Sonoff-Tasmota Releases page, and copy it to static/image_arduino.bin, replacing the existing file.", I do it but now....? Same process not work because I can not saw the itead-xxxx network....

sillyfrog, 2018-08-23:
@cpfarher That is just for the flashing stage using SonOTA, now you have Tasmota installed, you can follow the standard update process (as per that site). Good luck!

closed 2018-08-23 by sillyfrog