mirko/SonOTA #118: Sonoff Pow Module, not able to upgrade to latest Tasmota version

tali1974, 2018-04-07:
Hi.. Thanks for the script you have made. I was able to flash my POW over the air,
I have now Program Version | 5.9.1 installed. Now i wanted to upgrade to v5.12.0.

I go to "Firmware Upgrade" added this in the OTA Url
https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff-minimal.bin
and 
https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff.bin
but the box boot after some sec and come up with 5.9.1.. 

Any idea what iam doing wrong ?

LEMD49, 2018-04-11:
same issue here w POW

sillyfrog, 2018-05-07:
The firmware only supports upgrading via HTTP, see this page for more information: https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade

closed 2018-05-07 by sillyfrog