mirko/SonOTA #119: Sonoff Pow FW 2.04

zkzkzk2015, 2018-04-12:
- Operating System: Ubuntu 17.10
- Python Version: 3.5
- Sonoff model: Sonoff POW ver 2.0

trying to upload  tasmota via sonOTA but after switching to ITEAD ssid it asks me for the password. It supposed to be 12345678.
It is not accepting it. Any ideas?

Rapejim, 2018-05-12:
Try to use the flags:
`--legacy --slowstream`
(only legacy or both).

sillyfrog, 2018-05-17:
@zkzkzk2015 To clarify, you can't actually connect to the default AP from the Sonoff? If 12345678 is not working, I would suggest checking the documentation that came with it, as this initial stage is all standard (and would be required for the app).

zkzkzk2015, 2018-05-17:
Thanks for your replies, but unfortunately I changed my strategy and moved from OTA to cable.

bulletprooffool, 2018-05-17:
Given Sonoff’s reluctance to reply to this issue, perhaps we should target different hardware for this firmware project?
Hit them where it hurts?