mirko/SonOTA #120: Sonoff S22 stuck at finalStage (can I resume)

prankousky, 2018-04-14:
Hi,

I tried using SonOTA on a Sonoff S22 device.

- Operating System: macOS High Sierra v 10.13.3
- Python Version: 3.6.4
- Sonoff model: S22

Should have made sure that this device was compatible **before** attempting to flash it, but didn't :/ 

Everything worked until the finalStage wifi came up. Then it stopped working. I ended up unplugging the device and plugging it back in.

The finalStage AP still appears, but it has been a few days (didn't have time to report it before now). Is there a way I can resume flashing the device? Obviously, the *itead-xxxxxxx* AP is not there anymore because finalStage is there, so the python program will not let me re-flash the device. 

Is there some way I can get this flashed OTA?

Thanks in advance for your support :)

ftelleria, 2018-04-25:
Same issue here, I have a T1 R2 US

ftelleria, 2018-04-25:
Hi,

I've solved the issue, connect to the final stage WIFI and run the program with the option --no-prov:

python sonota.py --no-prov

It should resume the firmware update.

sillyfrog, 2018-05-07:
What @ftelleria said is correct, you can resume at any step, providing the previous step completed.

closed 2018-05-07 by sillyfrog