mirko/SonOTA #121: Sonoff basic stuck at 'Finalstage'

PjotrekSE, 2018-05-02:
- Operating System: Windoze 10 Home
- Python Version:  3.6.5
- Sonoff model:  basic V1.55
[debug_1524956542.log](https://github.com/mirko/SonOTA/files/1967814/debug_1524956542.log)
[debug_1524953751.log](https://github.com/mirko/SonOTA/files/1967836/debug_1524953751.log)

All works well until the final stage, after a while the console output says:
No longer on “FinalStage” SSID, all done!
Now connect to the sonoff-#### SSID and configure for your WiFi (it will not be configured).
Quitting.

BUT: The SSID ‘Finalstage’ does not go away, also not after rebooting the Sonoff.
Reran the script twice with the suggested option '--no-prov' to bypass the first stage, same result.
What to do?

Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.

giovannigenna, 2018-08-21:
@PjotrekSE Have u resolved ?

HFsi, 2019-02-09:
Same here on Ubuntu, FinalStage never shows up.
It´s no my first flashing with SonOTA, I´m doing the same thing like the other times, but I´m getting stuck on FinalStage, any ideas?

robross0606, 2019-11-25:
Similar problem here.  Several of these have worked fine.  Did the same thing on another and it never got past "FinalStage".  No errors, just hung there .  Eventually tried bouncing the Sonoff Basic after about 30 minutes of waiting.  It went right back to broadcasting the "FinalStage" SSID.  I can connect to the SSID, but nothing happens in sonOTA.  I also can't seem to get the device back to a stage 1 phase.  Looks like it might be bricked :'(

[debug_1574659077.log](https://github.com/mirko/SonOTA/files/3884965/debug_1574659077.log)

vakaobr, 2019-12-15:
Same issue here trying to sonOTA a basic unit, using MacOS, FinalStage ssid never shows up.

Kuki77, 2020-05-14:
Here also the same issue. The SSID FinalStage is not appearing after a the first stage.