mirko/SonOTA #133: Cannot Start SonOTA fine.

ettemito04, 2018-06-15:
- Operating System: Win 10
- Python Version:  3.6.5
- Sonoff model: Basic

C:\Users\ettor\OneDrive\Desktop\SonOTA-master>python sonota.py
Current IPs: ['192.168.0.250', '10.10.7.2']
Select IP address of the WiFi interface:
    0: 192.168.0.250
    1: 10.10.7.2
Select IP address [0]:
It looks like you are already on a Sonoff/Final stage WiFi network, please change to your normal WiFi network. If this has already been done, you may need to modify the IP range of your LAN to use this tool safely. If this is what you intended run again with --no-prov to skip this step.
Quitting.

C:\Users\ettor\OneDrive\Desktop\SonOTA-master>python sonota.py
Current IPs: ['192.168.0.250', '192.168.0.125']
Select IP address of the WiFi interface:
    0: 192.168.0.250
    1: 192.168.0.125
Select IP address [0]:
WiFi SSID: Vodafone-CasaPaneMonti
WiFi Password: *Censured*

Using the following configuration:
        Server IP Address: 192.168.0.250
        WiFi SSID: Vodafone-CasaPaneMonti
        WiFi Password: ************************
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
................Current IPs: ['192.168.0.250', '10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10003d3af0",
    "apikey": "314905d7-c204-4204-9a59-f36580c71309",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "Vodafone-CasaPaneMonti",
    "password": "************************",
    "serverName": "192.168.0.250",
    "port": 8443
}
Traceback (most recent call last):
  File "sonota.py", line 734, in <module>
    main()
  File "sonota.py", line 729, in main
    stage1()
  File "sonota.py", line 612, in stage1
    "http://10.10.7.1/ap", "POST", json.dumps(data))
  File "C:\Users\ettor\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1514, in request
    (response, content) = self._request(conn, authority, uri, request_uri, method, body, headers, redirections, cachekey)
  File "C:\Users\ettor\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1264, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "C:\Users\ettor\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1217, in _conn_request
    response = conn.getresponse()
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 258, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

i have also tried to change my IP:

Current IPs: ['192.168.0.251']
Select IP address of the WiFi interface:
    0: 192.168.0.251
Select IP address [0]:
WiFi SSID: Vodafone-CasaPaneMonti
WiFi Password: *CENSURED*

Using the following configuration:
        Server IP Address: 192.168.0.251
        WiFi SSID: Vodafone-CasaPaneMonti
        WiFi Password: ************************
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
..............Current IPs: ['192.168.0.251', '10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10003d3af0",
    "apikey": "314905d7-c204-4204-9a59-f36580c71309",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "Vodafone-CasaPaneMonti",
    "password": "************************",
    "serverName": "192.168.0.251",
    "port": 8443
}
Traceback (most recent call last):
  File "sonota.py", line 734, in <module>
    main()
  File "sonota.py", line 729, in main
    stage1()
  File "sonota.py", line 612, in stage1
    "http://10.10.7.1/ap", "POST", json.dumps(data))
  File "C:\Users\ettor\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1514, in request
    (response, content) = self._request(conn, authority, uri, request_uri, method, body, headers, redirections, cachekey)
  File "C:\Users\ettor\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1264, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "C:\Users\ettor\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1217, in _conn_request
    response = conn.getresponse()
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 258, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\ettor\AppData\Local\Programs\Python\Python36\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

ettemito04, 2018-06-15:
[debug_1529094471.log](https://github.com/mirko/SonOTA/files/2107554/debug_1529094471.log)
[debug_1529094012.log](https://github.com/mirko/SonOTA/files/2107555/debug_1529094012.log)

turagabhupathi, 2018-06-21:
I am also stuck with the same issue if your problem solved can u please share
`Microsoft Windows [Version 10.0.17134.112]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\Users\TEMP>python C:\Users\TEMP\Desktop\sonoff\SonOTA-master\SonOTA-master\sonota.py
Current IPs: ['192.168.0.5']
Select IP address of the WiFi interface:
    0: 192.168.0.5
Select IP address [0]: 0
WiFi SSID: CENTILLION-ACT
WiFi Password: *******

Using the following configuration:
        Server IP Address: 192.168.0.5
        WiFi SSID: CENTILLION-ACT
        WiFi Password: ******************
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
......Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10003c1802",
    "apikey": "625c1905-6881-4615-9ae1-504be3871210",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "CENTILLION-ACT",
    "password": "******************",
    "serverName": "192.168.0.5",
    "port": 8443
}
Traceback (most recent call last):
  File "C:\Users\TEMP\Desktop\sonoff\SonOTA-master\SonOTA-master\sonota.py", line 734, in <module>
    main()
  File "C:\Users\TEMP\Desktop\sonoff\SonOTA-master\SonOTA-master\sonota.py", line 729, in main
    stage1()
  File "C:\Users\TEMP\Desktop\sonoff\SonOTA-master\SonOTA-master\sonota.py", line 612, in stage1
    "http://10.10.7.1/ap", "POST", json.dumps(data))
  File "C:\Users\TEMP\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1514, in request
    (response, content) = self._request(conn, authority, uri, request_uri, method, body, headers, redirections, cachekey)
  File "C:\Users\TEMP\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1264, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "C:\Users\TEMP\AppData\Roaming\Python\Python36\site-packages\httplib2\__init__.py", line 1217, in _conn_request
    response = conn.getresponse()
  File "C:\Users\TEMP\AppData\Local\Programs\Python\Python36-32\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\TEMP\AppData\Local\Programs\Python\Python36-32\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\TEMP\AppData\Local\Programs\Python\Python36-32\lib\http\client.py", line 258, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\TEMP\AppData\Local\Programs\Python\Python36-32\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

C:\Users\TEMP>`