mirko/SonOTA #135: Stuck on Final Stage

turagabhupathi, 2018-07-03:
- Operating System: Windows 10
- Python Version: 3.6
- Sonoff model: Basic

Microsoft Windows [Version 10.0.17134.112]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\Users\TEMP>python C:\Users\TEMP\Desktop\SonOTA-master\SonOTA-master\sonota.py
Current IPs: ['192.168.0.17']
Select IP address of the WiFi interface:
    0: 192.168.0.17
Select IP address [0]: 0
WiFi SSID: CENTILLION-ACT
WiFi Password: **********

Using the following configuration:
        Server IP Address: 192.168.0.17
        WiFi SSID: CENTILLION-ACT
        WiFi Password: ******************
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
...............Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10003fb5f9",
    "apikey": "0b2812bc-ba8c-402c-8689-7ac3383310d4",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "CENTILLION-ACT",
    "password": "******************",
    "serverName": "192.168.0.17",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.0.17) is not assigned to any interface on this machine.
** Please change WiFi network to CENTILLION-ACT and make sure 192.168.0.17 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
......Current IPs: ['192.168.0.17']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Using selector: SelectSelector
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
........................

turagabhupathi, 2018-07-05:
Sorry My Sonoff is default 1.6 firmware and I successfully flashed manually

zomars, 2018-07-05:
I'm using the new T1 US 1C with the 1.8.3 firmware and I'm having this same issue.