mirko/SonOTA #136: Uncaught exception, closing connection.

tonygil, 2018-07-06:
Hi Have this  error , not sure what is causing it..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.....Uncaught exception, closing connection.
Traceback (most recent call last):
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 657, in _handle_events
    self._handle_read()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\platform\asyncio.py", line 117, in _handle_events
    handler_func(fileobj, events)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 657, in _handle_events
    self._handle_read()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
.....Uncaught exception, closing connection.
Traceback (most recent call last):
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 657, in _handle_events
    self._handle_read()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\platform\asyncio.py", line 117, in _handle_events
    handler_func(fileobj, events)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 657, in _handle_events
    self._handle_read()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
...............Uncaught exception, closing connection.
Traceback (most recent call last):
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 657, in _handle_events
    self._handle_read()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\platform\asyncio.py", line 117, in _handle_events
    handler_func(fileobj, events)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 657, in _handle_events
    self._handle_read()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "C:\Users\Tony\AppData\Roaming\Python\Python37\site-packages\tornado\iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "C:\Users\Tony\AppData\Local\Programs\Python\Python37\lib\ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
..Uncaught exception, closing connection.

sillyfrog, 2018-08-28:
As you are on Windows, please use Python 3.5, there is also the option of the prebuilt binary you can use.

There is some issue with 3.7 that I have not had time to look at.

closed 2018-08-28 by sillyfrog

thecableguy11492, 2018-12-11:
Hello, having a similar issue with Ubuntu 18.04, I've got a venv with python 3.6 installed.

[debug_1544498543.log](https://github.com/mirko/SonOTA/files/2665883/debug_1544498543.log)

brunetton, 2018-12-19:
Exactly the same error here, in a virtualenv using Python 3.5.6:

```
  File "/usr/lib/python3.5/ssl.py", line 641, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
....
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.............Uncaught exception, closing connection.
Traceback (most recent call last):
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/iostream.py", line 709, in _handle_events
    self._handle_read()
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/iostream.py", line 1581, in _handle_read
    self._do_ssl_handshake()
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/iostream.py", line 1501, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.5/ssl.py", line 996, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.5/ssl.py", line 641, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 127, in _run
    self._callback(*self._args)
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/platform/asyncio.py", line 122, in _handle_events
    handler_func(fileobj, events)
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/stack_context.py", line 300, in null_wrapper
    return fn(*args, **kwargs)
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/iostream.py", line 709, in _handle_events
    self._handle_read()
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/iostream.py", line 1581, in _handle_read
    self._do_ssl_handshake()
  File "/home/bruno/tmp/virtualenv/lib/python3.5/site-packages/tornado/iostream.py", line 1501, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.5/ssl.py", line 996, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.5/ssl.py", line 641, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
```

sillyfrog, 2018-12-19:
In all these cases I suspect it's also running newer firmware that does not work with SonOTA (as per the README.md), and Python does not like the way the Sonoff drops the connection.

brunetton, 2018-12-19:
I think https://github.com/mirko/SonOTA/issues/57 is the same issue as this one (this one is probably a duplicate)