mirko/SonOTA #138: POW R2 support

evlo, 2018-07-26:
- Operating System:  w10
- Python Version: 
- Sonoff model: POW R2

Can you please add support for r2?
My R2s are v 1.0, there is update for 1.1 in ewlink app, that I did not apply.

It seems that the /device /ap part in AP mode does work from the log, but device does not update firmware to tasmota

[debug_1532603407.log](https://github.com/mirko/SonOTA/files/2231760/debug_1532603407.log)

sillyfrog, 2018-07-30:
Looking at the logs, it appears to be another issue as per #58 - ie: all current firmware will not allow this to work any more. Annoyingly, they appear to now have 2 separate series of v1 firmware... :(

closed 2018-07-30 by sillyfrog

Mikepoplas, 2018-08-26:
Just yo update, 2.6.0 Firmware fail to flash too.

> Traceback (most recent call last):
>   File "/var/root/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 709, in _handle_events
>     self._handle_read()
>   File "/var/root/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1581, in _handle_read
>     self._do_ssl_handshake()
>   File "/var/root/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1501, in _do_ssl_handshake
>     self.socket.do_handshake()
>   File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 1108, in do_handshake
>     self._sslobj.do_handshake()
> OSError: [Errno 0] Error
> Exception in callback None()
> handle: <Handle cancelled>