mirko/SonOTA #139: Sonoff S31 Factory Firmware OTA?

meingraham, 2018-07-30:
- Sonoff model: S31
- ITead Firmware: 1.0.0

My sense is that the S31 firmware is different that then Sonoff Basic firmware and thus there is no correlation to the 1.6 version I have on the Basics I purchased directly from ITead in the same order.

I know that as of 1.6, ITead removed the ability to OTA update (although I've seen recent online noise that 2.0.4 may allow OTA again).

I can't quite seem to find a definitive answer for is whether I can OTA update an S31 with ITead 1.0.0 firmware. I found a few posts on several forums where it stated that SonOTA would not enter the final stage. But there was no confirmation that indeed it is not possible.

I'm just trying to get confirmation before I crack my units open that the only means to flash these devices is to upload new firmware via the serial interface.

Can anybody confirm?

Thanks.

Mike

sillyfrog, 2018-07-30:
Anything that I have received confirmation of, I try to always put on the wiki. My suggestion would be to try one, and see how it goes - worst case you have to crack it open to flash it anyway, best case it works :)

If it does work, if you can please update the wiki that would be great.

Cheers.

sillyfrog, 2018-07-30:
PS: Looking at #138 , it does not look like it will work unfortunately :(

closed 2018-07-30 by sillyfrog

meingraham, 2018-07-30:
Will do.

#138 was for a POW specifically... but it does appear, not surprisingly, that behaves as does the S31.