mirko/SonOTA #140: Stuck at Stage 2 - webserver not responding

dwarfer166, 2018-08-07:
- Operating System: Windows 10
- Python Version: 3.7.0
- Sonoff model: Basic (R2)

Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.
[debug_1533645565.log](https://github.com/mirko/SonOTA/files/2266395/debug_1533645565.log)

Sorry if this is already covered in another issue but the only one that was similar was #91 and it looks like OP didn't respond towards the end of the thread. 

I am trying to flash a Sonoff Basic with Sonota but I am having issues. Once my PC joins the wifi network of the Sonoff it then says to reconnect back to my usual network which I do. The sonoff connects to the network as well and I can see it gets an IP. But from here on it looks like it can never communicate with my PC. I am not convinced that the webserver is running properly. When I try to access the http/https pages I get a 404 error, is this correct? perhaps it is and I just be using a different URL. When I try to access the webpages the GET request does get logged in the Python window so that would lead me to think that it is working. 

When I wireshark traffic hitting the 8443 port but that's all. It really seems like my PC just isn't responding to the Sonoff at all. 
[image]

If anyone can provide any help it would be really appreciated! Thanks. 

Craig

sillyfrog, 2018-08-08:
Having a look at the logs, it looks like there are 2 IP's in the same subnet:

`2018-08-07 13:40:20,142: DEBUG: Current IPs: ['192.168.0.12', '192.168.0.41']`

I would suggest disabling one of these interfaces (eg: as a guess unplug the wired interface), and see i that helps. I would also suggest testing from another device on the network to check you can actually connect (if not, it could be a firewall or something).

Good luck.

sillyfrog, 2018-08-08:
The 404 is normal as there's nothing at the root address (rather it's files at specific URL's).

If you have not already, I would try to access the web interface from another device on the network and see if you get a response. eg: from your NAS something like `curl -vk https://192.168.0.41:8443/` - and ensure it can connect (getting a 404 is a success in this case).

dwarfer166, 2018-08-09:
That is good to know that the 404 error is normal. 

You know what, I should have thought about accessing the IP/port from a different machine on the network. Stupid. When I tried this I get no response despite it being accessible on the local machine. It doesn't make much sense to me though as the firewall is disabled and I am obviously accessing via the 192.168.0.41 IP and not 127.0.0.1 locally. Hmm. I have a wifi dongle to try with my NAS so I will see if I get the same behavior there. 

Any ideas why the IP might not be accessible from another machine? Would the webserver that is setup be configured to only allow connections from a certain IP address?

I appreciated your help sillyfrog! Thanks.

sillyfrog, 2018-08-09:
Windows can have a number of places for the firewall - it's a PITA. I would suggest looking at AV products, and ensuring your interface is set to a "home" or "work" network. The web server binds to the default (0.0.0.0) address, so it should be accessible from any interface at any time. 

I'll close this off for now, but feel free to re-open if you have further issues on this.

Good luck!

closed 2018-08-09 by sillyfrog

dwarfer166, 2018-08-14:
I just thought I would add to this if anyone else has the same issues as I have found the cause. 

So my primary machine for trying this on was Windows 10. After sillyfrog suggested checking the IP I found I could not access the IP:port in a browser from my other machine even though the firewall was off. When I tried the same thing on my Server 2012 NAS I found I could access the IP from my original IP.  I managed to get the Sonoff flashed with the initial Stage from my Server 2012 box but when I was on the 'FinalStage' SSID part I could not get it to work (I think this was probably caused by a mismatched port in retrospect). 

I had to go back to try and solve why my Windows 10 box IP:port wasn't accessible on the network. I literally had to Google 'Windows 10 hidden firewall' as that is what it felt like but this led me to 'Windows Defender Security Centre' where there seems to be a whole other firewall to disable. Weirdly this was only active on the 'Domain' network and my network was being categorised as a 'Private' network. Nevertheless, I disabled this and BINGO, straight away I could access IP:port from other machines. 

I wasn't out of the woods yet though, it still wasn't working properly, I had to wireshark the connection where I found that the Sonoff was trying to connect via port 8080 even though I had set it in the sonota.py file to be 8081. I think what had happened is that I must have ran through Stage1 on my NAS box with the default port before changing it to 8081. Obviously whatever port you perform Stage1 with needs to still be accessible when in Stage2 as that is all the Sonoff knows. 

Anyways, tips to note:
1. If using Windows 10 make sure the firewall in Windows Defender Security Centre is disabled.
2. Wireshark to get more info!

Thanks.