mirko/SonOTA #143: 4CH Pro R2 v 2.0.1 stuck on FinalStage

giovannigenna, 2018-08-21:
- Operating System: Mac OSX High Sierra 10.13.6
- Python Version: 2.7.10
- Sonoff model: 4CH Pro R2 v 2.0.1
[debug_1534869870.log](https://github.com/mirko/SonOTA/files/2307667/debug_1534869870.log)
Hi,
I'm trying to flash my Sonoff 4CH Pro R2 v 2.0.1
I have done all steps, but now the Sonoff looks bricked. I can see the Wifi network "FinalStage" and I can connect to it and get the IP 192.168.4.2. The first 3 leds are red and on and the blue led is on too. 
I tried many times to restart the py script, but it look like that the sonoff device won't connect it to download the firmware.... I'm also checked if the HTTP server on port 8080 was up and running trying to download the image from another device...

How I can fix it ?

giovannigenna, 2018-08-21:
I also try with Windows 10, python 3.5.4 but I get the same problem.
I see that when I restart the son off the blue led blink slowly like 4/5secs, when a pc connect to its wifi and start the python script it stop blink and is on, but no request arrived to HTTP server :/

Lodeli, 2018-08-21:
Same problem here but using sonota.exe
Can connect to FinalStage and get the ip address; also 192.168.4.1 is the default gw and I can ping it. 
But nothing is listening on that address and physical buttons are just not working.

Using --no-prov, when connecting to FinalStage it keeps starting over and over uploading with no errors but also without completing.

sillyfrog, 2018-08-21:
If sonota is running correctly, it should be running a web server, and the FinalStage program on the Sonoff should be connecting to that web server to download the final image. I would triple check you have all firewalls disabled. This is a new SSID, so it'll have it's own configuration. I would also suggest running a ```tcpdump``` or wireshark, and seeing if the Sonoff is trying to connect in (should be every minute or so). Also ensure your OS is not dropping the connection to go to another WiFi network as there is no internet available.

The log you have there shows the start of things going as they should (```DEBUG: Sending file: /ota/image_arduino.bin```). So it's always worth a full power cycle of the Sonoff as well.

Lodeli, 2018-08-22:
Just to complete, as previously I wasn't able to attach files, here is my last log using **sonota.exe --no-prov** and connecting to FinalStage.
[debug_1534881233.log](https://github.com/mirko/SonOTA/files/2309303/debug_1534881233.log)

Once at home, I'll try checking the firewall again and trying downloading the file from the sonota.exe web server using a different device, though I'm quite sure the firewall was completely disabled for all kinds of LANs (FinalStage was also set as "private").

giovannigenna, 2018-08-22:
Hi @sillyfrog ,
I have also tried with Linux with firewall disabled and I get the same result.
I also see that while I'm not running the py scripts the sonoff blink slowly, like 5 secs on and 5 secs off. When it start the py scripts it stop blinking, but no request arrives....
I also try with the `--slowstream `and `--no-prov` params...

If I cannot flash by OTA I can also flash by serial port, right ? it is not bricked, right ?
If I can flash it by serial port, have you some guide to follow ?

Thanks

sillyfrog, 2018-08-23:
That's very weird as the firmware has obviously uploaded (mostly) OK, but something is not running like it should (it's very simple, so not sure what it could be). If you take a tcpdump, I can try looking at the output and seeing if I can debug from there.

Yes, even if the OTA process completely fails, you can still flash via serial.

If you are going down that path, it would be good to get the serial output while trying to flash (that's where the debugging output is sent by Final Stage), to try and figure out what's going on for others.

Cheers.

Lodeli, 2018-08-23:
@sillyfrog I successfully flashed the device using a bare Arduino UNO as USB->Serial; sorry, too early to see your comment and get the serial output. Anyway, I can confirm the firewall was absolutely disabled also on FinalStage specific LAN.
Looking around I'v seen I'm not the only one with this kind of problem and that you've been thinking about some flash issue.

@giovannigenna I flashed with no problems but this: the [guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro) says "TX/RX are printed correctly on the pro version", that was not my case, I had to connect TX->TX and RX->RX to finally get it working, don't know why.

giovannigenna, 2018-08-24:
@Lodeli thanks,

giovannigenna, 2018-08-24:
@sillyfrog I have connect my board to serial port and I get this log while try to flash from FinalStage:
```
...........................Done
Current rom: 2
Rom 1 magic byte: 0xE9
Reflashing rom: 0
Configuring Wifi FinalStage...
Waiting for connection...
.Waiting for connection...
.Connected
Web server accessible...
Flash Mode: 3
ROM or Flash Mode not recognized 
```
I will try to flash it by serial port...
@Lodeli thanks about your feedback, but in my board to connect the serial port I have connect TX->RX and RX ->TX.

I'll give you some feedback after serial flash ;)

Lodeli, 2018-08-24:
@giovannigenna Then mayby that was something related to the Arduino board and not the Sonoff one

sillyfrog, 2018-08-24:
@giovannigenna Thanks! That gives me something to work with. It appears the PRO has a different flash type to other devices (at least some versions). Not something I have seen before, but at least something I can look at to try and resolve moving forwards.

giovannigenna, 2018-08-24:
I confirm that I had flashed my 4CH PRO r2 by serial port. I followed this [guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro) and used  [VS + PlatformIO](https://github.com/arendst/Sonoff-Tasmota/wiki/PlatformIO) with Arduino IDE I got some errors while trying to compile it.
I can confirm that in my board RX and TX are correct, so I had connected TX->RX and RX->TX.
For the serial port I used [Pixnor FT232RL](https://www.amazon.it/gp/product/B013QRUA6W/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)

@sillyfrog I suggest to remove the 4CH PRO r2 v2.0.1 from supported device while this bug is not fixed. Also I would like to thank you for the support

tapesandrecords, 2018-08-26:
@giovannigenna I just purchased the 4CH PRO r2 and have been trying to flash it to no success.  

I've tried two different methods:

- [Flashing via RPI](https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi)
- Flashing via the [guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro) from the link you posted. Tho I wonder if I'm not using the right usb/serial dongle in this particular case.

Both times I'm not able to do it. I also tried OTA, but never got past Final Stage.

Hoping for any additional insight on how you did it?

giovannigenna, 2018-08-27:
Hi @tapesandrecords,
Even if I had a Raspberry and Arduino I prefered buy a USB -> Serial adapter.
I suggest you to buy one like mine and test it again. With VS + PlatformIO, that was really simple to do.

TmShaz, 2018-08-28:
Exactly the same problem here, like giovannigenna described in first post. Will purchase a serial adapter now and try flash by serial port. Hope this one will work for me: https://www.amazon.de/dp/B01C2P9GD2/ref=cm_sw_em_r_mt_dp_U_NrqHBb5Z2054X

giovannigenna, 2018-08-28:
@TmShaz emember to set the 3.3 volt on your USB->Serial. Usually are configured to use 5v in default mode. Just move the black jumper near the serial pins

tapesandrecords, 2018-08-29:
Well, I've tried with these two usb-serial adapters, this [one](https://www.amazon.com/gp/product/B014Y1IMNM) and this [one](https://www.amazon.com/gp/product/B00N4MCS1A/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1). And I'm unable to get it to work.

I've even tried pairing with the [native app](http://sonoff.itead.cc/en/ewelink) and that doesn't even work. And it looks like [other people](http://support.iteadstudio.com/support/discussions/topics/11000011929/page/1?url_locale=) have had problems with that as well.

TmShaz, 2018-08-30:
Couldn't wait longer for my usb-serial adapter and tried with my Raspberry Pi3. Worked. :)
I used the [Flashing via RPI](https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi) user guide. First i was a little bit irritated by the picture shown on the special side for [4CH Pro R2](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro) because it looks like you have to use the STM Programmer for TX. But no, just use the GRD, TX and RX Pins of ESP programmer. In my case i used a 12V 5A power supply and didn't use the 3,3 Pin. Do not connect the 3,3V Pin at the same time with external Power.
[image]

Many thanks to all helpers

TmShaz, 2018-09-07:
After a few days of testing module running pretty good but the buttons doesn't work as described here: [button usage](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage)

sillyfrog, 2018-09-09:
@giovannigenna FYI: I have made some minor changes to try and catch this case you came across with the 4CH Pro. As I don't have one and can't actually test it, I can't guarantee it'll work, but hopefully it should be good now for others moving forwards.

I'll close this ticket, and we can re-open it should the same thing happen again.

closed 2018-09-09 by sillyfrog