mirko/SonOTA #151: No FinalStage SSID- RF Bridge 433 - Firmware 1.0

GuyFromMars365, 2018-09-23:
- Operating System: Win 10
- Python Version: Unknown (Using SonOTA.exe)
- Sonoff model: RF Bridge 433
- Sonoff firmware version: 1.0
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware: 

[debug_1537726190.log](https://github.com/mirko/SonOTA/files/2409196/debug_1537726190.log)

Tried doing this a number of times, with mostly the same results: No FinalStage SSID to connect with. The first time I forgot to turn off the firewall and I got a "WARNING: 404 GET /" on stage 2, but no FinalStage SSID to connect with. In another instance,  SonOTA.exe closed on it's own after connecting to SSID ITEADxxxxx. In each case, when repeating the procedure, I simply disconnect the Sonoff power, reconnect. Then rerun the .exe, hold the Pairing Button down for about 7 seconds, release, then hold another 7 seconds to put it in AP mode. I then connect when the .exe tells me to and always gets stuck on Stage 2, because there is never a FInalStage SSID to connect with. :/ I'm wondering if I have to resort to a FTDI cable now (never done before). :/

WoosterInitiative, 2018-10-05:
I'm having a similar problem, though after the first attempt, power-cycling doesn't seem to help and my logs are a little different.
[debug_1538717593.log](https://github.com/mirko/SonOTA/files/2449059/debug_1538717593.log) (all recent attempts)
[debug_1538716303.log](https://github.com/mirko/SonOTA/files/2449060/debug_1538716303.log) (most successful attempt)

sillyfrog, 2018-10-07:
I'll update the README to reflect this, however these logs show the known issue with the current releases of the Sonoff - ITEAD is actually doing some security checks now, and it's no longer possible to hijack this mechanism to put on custom firmware - if you can post to ITEAD, that may assist in getting it across the line (but I'm not holding my breath).

closed 2018-10-07 by sillyfrog