mirko/SonOTA #153: Behaviour of Sonof Touch with firmware 1.8.1 (hope it helps for future developments)

bubibubi, 2018-10-01:
I'm flashing some Sonoff Touch and Sonoff T1 (1/2/3 gangs) using SonOTA.
As expected on some Sonoff it works, on other Sonoff does not work.
About not working firmwares, I'm trying to flash Sonoff Touch with 1.8.1 firmware (the latest for the Sonoff Touch).
On Sonoff Touch 1.8.1 firmware I've seen a strange behaviour:
**mode 1.** If I press the button for 7 seconds the blinks behaviour is short short long pause short short long
**mode 2.** In mode 1, if I press the button for 7 seconds it blinks short short short short ...

For flash with SonOTA I need to put the device in mode 2 otherwise I can't see the ITEAD-xxxxx AP.
SonOTA can set the device
```
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100021443f",
    "apikey": "0c694274-456d-4539-9c9b-7b9d155ec96d",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "password": "********",
    "serverName": "10.0.0.216",
    "port": 8443,
    "version": 4,
    "ssid": "MyAP"
}
```

then I fall back the server to MyAP and I can see (using wireshark) that the device speaks with my server (a connection about every 60 seconds) but the update does not start (so no FinalStage AP).

**Behaviour with eWeLink software**
I've tried to couple the same device with eWeLink software.
If I go in mode 1 (With a network analyzer I can't see any AP appearing) eWeLink writes "Found second generation device" and it can configure it
If I go in mode 2 (With a network analyzer I see the ITEAD-xxxx AP appearing) eWeLink writes "Found first generation device" and it can't configure it (2 minutes timeout).

sillyfrog, 2018-10-07:
This is a known issue as SonOTA was exploiting the lack of security on the Sonoff, which ITEAD have now fixed. Please see the README for more information.

closed 2018-10-07 by sillyfrog

bubibubi, 2018-10-07:
(I'm asking this question for fixing SonOTA not for flashing my hardware).
Actually I can't understand (immagine) the procedure that the eWeLink app uses to couple the switch with my wifi network using this new system (no AP).
Could it be an hidden AP?
Do you have any docs about it?

sillyfrog, 2018-10-07:
Regarding the different modes (holding for 7 seconds, then doing it again), I have seen this as well and not looked too hard at it, I figured it was probably a bug in the Sonoff.

Everything that's been done for SonOTA has been by reverse engineering the network traffic and figuring it out from there. I have sent support tickets to ITEAD, and they appear friendly, but then they stop responding when it gets to anything "real", like helping getting projects like this going :(

csanz91, 2019-03-07:
I'm very curious about how the **mode 1** works. ¿Has anyone found out how it works? 
I can't see any hidden networks created by the device and the phone doesn't disconnect from my home network during the setup process ¿how the hell is it sending the wifi credentials to the device?