mirko/SonOTA #156: questions about T1 us 3 ch

neomvll, 2018-11-17:
- Operating System: w10
- Python Version: 3.6.4
- Sonoff model: T1 US 3 CH
- Sonoff firmware version: 2.0.1
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware: Yes
Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.

My T1 US come today, but I run to put it in the wall, and now all 3 lights is working like a charm, but I have 2 sonoff RF w/ 2.6 and the connection w/ china servers is a shit. So I want flash my T1(to use on my HassOS) over the air and my questions are: 
- Can I flash my T1 when it is installed(and working) or I need desconnect all lights to do this??
- May I do a backup from original firmware? If yes, how?
Thanks

sillyfrog, 2018-11-17:
Glad it worked on one device.

Given it v2.6 firmware, OTA updates won't work (see home page).

To backup the firmware, you'll need to connect serial anyway, see here for how to run a backup via serial. Note you'll need to do this for each device as they firmware is hard coded for the MAC (or similar) on the device: https://github.com/mirko/SonOTA/issues/1#issuecomment-334724811

closed 2018-11-17 by sillyfrog

neomvll, 2018-11-17:
> Glad it worked on one device.
> 
> Given it v2.6 firmware, OTA updates won't work (see home page).
> 
> To backup the firmware, you'll need to connect serial anyway, see here for how to run a backup via serial. Note you'll need to do this for each device as they firmware is hard coded for the MAC (or similar) on the device: [#1 (comment)](https://github.com/mirko/SonOTA/issues/1#issuecomment-334724811)

Thanks for replay :)
The 2.6 firmware is on my sonoff RF, so I lost it, but my question is about T1 with 2.0.1:
Can I flash over the air my T1 when it is installed(and working in my wall) or I need desconnect all lights to do this??

sillyfrog, 2018-11-17:
You should be able to do it with the lights connected - they'll likely flash a few times during the process - but the actual firmware stage won't care if something is connected or not (as it's just controlling a relay).

Good luck!