mirko/SonOTA #157: Why is WIFI_SSID needed?

wewa00, 2018-11-21:
"--wifi-ssid WIFI_SSID
The ESSID of the WiFi network the device should eventually connect to."

I actually don't see any reason why the Sonoff device should need connect to the home WIFI.
Which eventuality is meant here?

sillyfrog, 2018-11-21:
That's how the Sonoff downloads the new firmware, SonOTA tells the Sonoff "connect to this WiFi, and download updates from here" (where here is a web server running on your PC).

wewa00, 2018-11-22:
Ok, I thought for that the ITEAD-* network is used.

Are the different files (image_user1-0x01000.bin, ...) downloaded using different networks?

sillyfrog, 2018-11-22:
Check out the sections under https://github.com/mirko/SonOTA#update-steps - this outlines what each bit is doing, and further down some more details.

closed 2018-11-22 by sillyfrog