mirko/SonOTA #158: sonota.exe vs sonoff TH10

neomvll, 2018-11-26:
- Operating System: win 10
- Python Version: 
- Sonoff model: TH10
- Sonoff firmware version: 2.0.4
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware: 

Hi guys, Is sonota.exe working in sonoff TH10 2.0.4? I know its listed, but was flashed over the air??

thanks :)

Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.

neomvll, 2018-11-29:
[image]

final stage never comes in firmware 2.0.4, but its listed :/
tried with --legacy, --slowstream, no works.

pererikl, 2018-12-16:
It worked ota for me.
Shut down all firewalls! I forgot the last on windows firewall for public networks like five times...
Windows firewall needs to be turned off on three places.

neomvll, 2018-12-16:
> It worked ota for me. fw 2.04 on TH10
> Shut down all firewalls! I forgot the last one, windows firewall for public networks, like five times...
> Windows firewall needs to be turned off on three places.

well, not worked here, and as I said, I flashed others sonoffs before, so firewalls and stuff was right. But tasmota by FTDI works, so maybe was my TH10, I don't know. At least its not ITEAD software anymore haha