mirko/SonOTA #162: Sonoff S20 firmware: 1.8.1

leonard88, 2019-01-07:
- Sonoff model: S20
- Sonoff firmware version: 1.8.1
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware: 

Hi

Do someone have any experience upgrading 1.8.1 original firmware with SonOTA? Is it possible to upgrade S20 1.8.1 with SonOTA at all?

Thansk!

Piega, 2019-02-01:
This is a list of my tests:

Original Firmware version -- if is upgradable with sonota -- sonoff model

2.0.1 yes ( T1 us )
2.0.4 yes (TH16)
2.6.0 no ( 4ch pro R2, T1 us )
2.6.1 no ( POWR2 )
2.7.0 no ( 4ch pro R2, T1 us )
2.7.1 no ( 4ch pro R2 )
2.8.0 no ( POWR2 )

eloo, 2019-03-18:
Looks like there is a new major version (3.0.0) has anybody tested it already?

eloo, 2019-03-20:
okay.. just tested it.. seems still not to work out of the box

yarosman, 2019-05-01:
> Looks like there is a new major version (3.0.0) has anybody tested it already?

I got sonoff with 3.0.1 and it doesn't work (

CarlosGS, 2020-05-03:
Nope, not working with the latest versions. I understand the only option is to solder cables and flash with a cable?