mirko/SonOTA #164: ESP8285 based relay board doesn't like certificate

therealbat, 2019-02-03:
- Operating System: macOS mojave (10.14)
- Python Version: 3.6.8
- Sonoff model: DM-581 (DIYMORE, see link below)
- Sonoff firmware version: 1.7.0
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware:  yes

link to the device: https://www.diymore.cc/products/diymore-dc12v-esp8285-self-lock-smart-home-wifi-wireless-switch-delay-relay-module-by-app-control-ios-android

process get stuck at stage2, the device is periodically trying to contact tornado (tried both 8443 and legacy) but the connection is FIN,ACK-ed after a few packets. did a packet capture, and it seems the stuff goes off right after server hello.  with legacy the device returns to its original working state after two tries, with port 8443 it tries to update forever.

therealbat, 2019-02-03:
(pressed enter accidentally :-)

therealbat, 2019-02-03:
[image]

therealbat, 2019-02-03:
using charles i was able to capture the communication between the EWelink app and the OTA server:

bibiak1, 2019-04-10:
Hi,

I have same issue. Tried on Mac, on Windows (sonota.exe shutdown unexpectedly), on docker.
Is it possible that it's related to cypher type on https communication?
Most openssl versions I saw doesn't support SSL2/SSL3 encryption.

NdK73, 2019-07-13:
From SonOff fw 1.6 SonOTA does not work anymore.
Started a Twitter campaign to give a signal to Itead.
https://twitter.com/NdK_BO/status/1150142117282549760
Retweet at will :)