mirko/SonOTA #168: [New Smart G2 2.0.1]Successfull flash, help for template/model selection

FrancYescO, 2019-06-10:
Successfully flashed this device and upgraded via OTA (http://thehackbox.org/tasmota/release/sonoff.bin) to 6.5.0
[image]
Bought from here (90-250V **no-**RF module):
https://www.aliexpress.com/item/220v-inching-selflock-interlock-smart-relay-switch-220v-wifi-module-RF433-app-control-wifi-interruptor-alexa/32933270623.html?spm=a2g0s.9042311.0.0.27424c4dbwVMH3

But actually i'm not able to configure the model/template, if i press the first button one of the relays wil start switching in a loop, the other 2 seems to do nothing... do you have a template

FrancYescO, 2019-06-14:
This is the right template https://blakadder.github.io/templates/geekcreit_2ch.html
Be sure to read the ***note*** in that page.

closed 2019-06-14 by FrancYescO

gnasser76, 2019-12-11:
hello i bought this. https://www.aliexpress.com/item/32996254506.html?spm=a2g0s.9042311.0.0.27424c4dBDIrzN

is there a tasmota firmware as well as template ?