mirko/SonOTA #169: set_nodelay?

altontoth, 2019-07-10:
Thoughts? Never get FinalStage. Just loops broken like this.

2019-07-10 00:02:22,683: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2019-07-10 00:02:22,683: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2019-07-10 00:02:22,684: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2019-07-10 00:02:31,316: DEBUG: << HTTP POST /dispatch/device
2019-07-10 00:02:31,316: DEBUG: >> /dispatch/device
2019-07-10 00:02:31,320: DEBUG: >> {
2019-07-10 00:02:31,321: DEBUG:     "error": 0,
2019-07-10 00:02:31,322: DEBUG:     "reason": "ok",
2019-07-10 00:02:31,324: DEBUG:     "IP": "192.168.1.72",
2019-07-10 00:02:31,326: DEBUG:     "port": 8443
2019-07-10 00:02:31,330: DEBUG: }
2019-07-10 00:02:31,330: INFO: 200 POST /dispatch/device (192.168.1.73) 14.99ms
2019-07-10 00:02:31,658: INFO: 101 GET /api/ws (192.168.1.73) 0.00ms
2019-07-10 00:02:31,659: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:31,659: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:34,760: INFO: 101 GET /api/ws (192.168.1.73) 0.00ms
2019-07-10 00:02:34,760: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:34,761: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:39,001: INFO: 101 GET /api/ws (192.168.1.73) 2.03ms
2019-07-10 00:02:39,002: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:39,005: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:41,572: INFO: 101 GET /api/ws (192.168.1.73) 0.91ms
2019-07-10 00:02:41,572: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:41,573: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:45,767: INFO: 101 GET /api/ws (192.168.1.73) 0.90ms
2019-07-10 00:02:45,767: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:45,768: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:48,964: INFO: 101 GET /api/ws (192.168.1.73) 0.00ms
2019-07-10 00:02:48,964: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:48,965: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:53,161: INFO: 101 GET /api/ws (192.168.1.73) 1.00ms
2019-07-10 00:02:53,161: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:53,162: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:56,354: INFO: 101 GET /api/ws (192.168.1.73) 0.00ms
2019-07-10 00:02:56,355: DEBUG: << WEBSOCKET OPEN
2019-07-10 00:02:56,356: ERROR: Uncaught exception GET /api/ws (192.168.1.73)
HTTPServerRequest(protocol='https', host='iotgo.iteadstudio.com', method='GET', uri='/api/ws', version='HTTP/1.1', remote_ip='192.168.1.73')
Traceback (most recent call last):
  File "C:\Users\Alton\AppData\Roaming\Python\Python37\site-packages\tornado\websocket.py", line 956, in _accept_connection
    open_result = handler.open(*handler.open_args, **handler.open_kwargs)
  File "B:\GitHub\sonota\sonota.py", line 194, in open
    self.stream.set_nodelay(True)
AttributeError: 'NoneType' object has no attribute 'set_nodelay'
2019-07-10 00:02:58,561: INFO: 101 GET /api/ws (192.168.1.73) 0.00ms

a5m0, 2019-08-18:
Also getting this error. Using sudo and --legacy on a sonoff th16 from ubuntu host.

minsuke, 2019-09-25:
I also same error did you solved this problem ?

dragosch, 2019-10-27:
comment out the line
#self.stream.set_nodelay(True)
this helped