mirko/SonOTA #172: Script crashed starting web server

arodriguezf, 2019-11-15:
- Operating System: Windows 10
- Python Version: python 3.8.0
- Sonoff model: sonoff basic
- Sonoff firmware version: not sure, never used an it has more than 1 year
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware: yes

** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
.........................Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "XXXXXXXX",
    "apikey": "XXXXXXXXXXXXXXXXXXXXXXXX",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "XXXXXXXX",
    "password": "*************",
    "serverName": "10.0.0.67",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (10.0.0.67) is not assigned to any interface on this machine.
** Please change WiFi network to SweetHome and make sure 10.0.0.67 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
...Current IPs: ['10.0.0.67']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Using proactor: IocpProactor
Traceback (most recent call last):
  File "C:\Repos\SonOTA\sonota.py", line 736, in <module>
    main()
  File "C:\Repos\SonOTA\sonota.py", line 732, in main
    stage2()
  File "C:\Repos\SonOTA\sonota.py", line 654, in stage2
    app.listen(DEFAULT_PORT_HTTP)
  File "C:\Users\arfer\AppData\Local\Programs\Python\Python38-32\lib\site-packages\tornado\web.py", line 2112, in listen
    server.listen(port, address)
  File "C:\Users\arfer\AppData\Local\Programs\Python\Python38-32\lib\site-packages\tornado\tcpserver.py", line 152, in listen
    self.add_sockets(sockets)
  File "C:\Users\arfer\AppData\Local\Programs\Python\Python38-32\lib\site-packages\tornado\tcpserver.py", line 165, in add_sockets
    self._handlers[sock.fileno()] = add_accept_handler(
  File "C:\Users\arfer\AppData\Local\Programs\Python\Python38-32\lib\site-packages\tornado\netutil.py", line 279, in add_accept_handler
    io_loop.add_handler(sock, accept_handler, IOLoop.READ)
  File "C:\Users\arfer\AppData\Local\Programs\Python\Python38-32\lib\site-packages\tornado\platform\asyncio.py", line 99, in add_handler
    self.asyncio_loop.add_reader(fd, self._handle_events, fd, IOLoop.READ)
  File "C:\Users\arfer\AppData\Local\Programs\Python\Python38-32\lib\asyncio\events.py", line 501, in add_reader
    raise NotImplementedError
NotImplementedError

After that, the script ends. Any idea?
[debug_1573845850.log](https://github.com/mirko/SonOTA/files/3852919/debug_1573845850.log)

theomythos, 2019-12-10:
Have you got around it?

arodriguezf, 2019-12-10:
No didn´t find a solution and didn´t have any response

arodriguezf, 2019-12-10:
> Have you got around it? Ran into the same issue with Sonoff Mini.

No didn´t find a solution and didn´t have any response

HasslerEngineering, 2020-02-29:
have the same Issue here. Any Updates?

kellerza, 2020-04-27:
#177