mirko/SonOTA #173: eWeLink Mini PCI-e Desktop PC Remote Control

gnasser76, 2019-12-11:
hello i bought this https://www.aliexpress.com/item/32996254506.html?spm=a2g0s.9042311.0.0.27424c4dBDIrzN 
its buggy and keeps disconnecting . is there a tasmota firmware for it ? and what about the template 

Thank you guys

nicandris, 2019-12-30:
I bought the same one. Tried flashing it several times, for hours but it doesn't work. It stays to the Finalstage stage and fails. I can see the device flashing in a way that means server connection error, so my guess it's because of the firmware not being compatible any more with sonOTA (there is a link in the readme asking for firmware downgrade)

gnasser76, 2019-12-30:
we need to ask for a custom firmware . i posted my problem on discord . and no one answered.

frankhommers, 2020-03-18:
Do you have any progress? I have bought this device in the USB powered version.

eaddictuk, 2020-11-07:
I Have flashed it with Tasmotizer