mirko/SonOTA #175: Sonota.exe not progressing after connecting to ITEAD-SSID

benziman88, 2020-03-28:
- Operating System: Win10 1909 x64
- Python Version: Sonota.exe
- Sonoff model: 4 ch pro R2
- Sonoff firmware version: ?

I start sonota.exe, enter my wifi credentials, connect to the ITEAD-SSID, some additional lines appear (no flashing though) but then Laptop disconnects from ITEAD-SSID and back to my normal wifi. If I try to reconnect to ITEAD-SSID get error connection not possible.

Using the following configuration:
        Server IP Address: 172.16.77.115
        WiFi SSID: WifiXY
        WiFi Password: *********
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
...............................Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "apikey": "a9a9bb5d-c2cd-4f48-83ac-74bb9a0c04bf",
    "accept": "post",
    "deviceid": "10009f1b79"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "password": "*********",
    "version": 4,
    "ssid": "WifiXY",
    "port": 8443,
    "serverName": "172.16.77.115"
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (172.16.77.115) is not assigned to any interface on this machine.
** Please change WiFi network to WifiXY and make sure 172.16.77.115 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
......Current IPs: ['172.16.77.115']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

buteomont, 2020-05-10:
I am having this same issue.  I never sends anything to the Sonoff after connecting, and the part that begins with  "*** IMPORTANT! ***" just repeats every minute or so.  Did you find a solution?

Kuki77, 2020-05-14:
Same issue here on a Debian machine and using a Sonoff Basic with old  FW (1.55).
No FinalStage SSID is established by the device after passing first stage.

atus42, 2020-06-12:
Unfortunately I have the same issue. :-(

pirracas77, 2020-08-06:
Same issue here. I wonder if the sonoff manages to connect to the ssid