mirko/SonOTA #182: Latest firmware

orrionis, 2020-09-28:
- Operating System: windows 10
- Python Version: 3.8
- Sonoff model: t1
- Sonoff firmware version: 3.4
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware: 

I know it's works with 1.6 but i was wondering since this time there is an issue. Or it s impossible,  latests firmware it s lock

sillyfrog, 2020-09-30:
Extremely unlikely it'll work given what ITead have done to lock it down. I personally don't use Sonoff's any more (for this reason), so the best best it to give it a go to test it (see the first couple of paragraphs in the README for the disclaimer).

closed 2020-09-30 by sillyfrog

Weissnix4711, 2020-12-15:
@sillyfrog Do you know which changes have been made to the firmware? `/device` still returns relevant info, although the device refuses to connect and perform the OTA update.

Could you give a better explanation as to what's been figured out so far, and whether OTA flashing later v2 or any v3 is at all feesable?

Thank you.

sillyfrog, 2020-12-21:
@Weissnix4711 Check out the start of the README and associated links. In short, they are doing SSL certificate verification, so not something we are going to be able to work around.

Weissnix4711, 2020-12-21:
Ah, I gotcha. But the question is.. Why hasn't Sonoff been doing this from the beginning lol.