mirko/SonOTA #184: Never ending loop at stage2 when trying to flash Tasmota for the first time

iminet, 2020-11-20:
- Operating System: Windows 10 x64
- Python Version: 
- Sonoff model: Sonoff Basic R2
- Sonoff firmware version: 3.5.0
- I have read the README, and understand that SonOTA does not work with the latest Sonoff firmware: 
[debug_1605859859.log](https://github.com/mirko/SonOTA/files/5572365/debug_1605859859.log)

Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.

Hi All, I've trired to flash Tasmota with SonOTA for the first time. Everything was fine until the program asked to change back to my wifi and stepped over to stage2. Well, at this point, nothing happened for minutes, minutes and more minutes.
What's wrong? May I get some support to solve this issue? Thanks!

The Powershell window contents can be seen below.

`PS C:\Users\keem1\Desktop\SonOTA-master> .\sonota.exe
[9332] PyInstaller Bootloader 3.x
[9332] LOADER: executable is C:\Users\keem1\Desktop\SonOTA-master\sonota.exe
[9332] LOADER: homepath is C:\Users\keem1\Desktop\SonOTA-master
[9332] LOADER: _MEIPASS2 is NULL
[9332] LOADER: archivename is C:\Users\keem1\Desktop\SonOTA-master\sonota.exe
[9332] LOADER: Extracting binaries
[9332] LOADER: Executing self as child
[9332] LOADER: set _MEIPASS2 to C:\Users\keem1\AppData\Local\Temp\_MEI93322
[9332] LOADER: Setting up to run child
[9332] LOADER: Creating child process
[9332] LOADER: Waiting for child process to finish...
[2296] PyInstaller Bootloader 3.x
[2296] LOADER: executable is C:\Users\keem1\Desktop\SonOTA-master\sonota.exe
[2296] LOADER: homepath is C:\Users\keem1\Desktop\SonOTA-master
[2296] LOADER: _MEIPASS2 is C:\Users\keem1\AppData\Local\Temp\_MEI93322
[2296] LOADER: archivename is C:\Users\keem1\Desktop\SonOTA-master\sonota.exe
[2296] LOADER: SetDllDirectory(C:\Users\keem1\AppData\Local\Temp\_MEI93322)
[2296] LOADER: Already in the child - running user's code.
[2296] LOADER: manifestpath: C:\Users\keem1\AppData\Local\Temp\_MEI93322\sonota.exe.manifest
[2296] LOADER: Activation context created
[2296] LOADER: Activation context activated
[2296] LOADER: Python library: C:\Users\keem1\AppData\Local\Temp\_MEI93322\python35.dll
[2296] LOADER: Loaded functions from Python library.
[2296] LOADER: Manipulating environment (sys.path, sys.prefix)
[2296] LOADER: Pre-init sys.path is C:\Users\keem1\AppData\Local\Temp\_MEI93322\base_library.zip;C:\Users\keem1\AppData\Local\Temp\_MEI93322
[2296] LOADER: sys.prefix is C:\Users\keem1\AppData\Local\Temp\_MEI93322
[2296] LOADER: Setting runtime options
[2296] LOADER: Bootloader option: pyi-windows-manifest-filename sonota.exe.manifest
[2296] LOADER: Initializing python
[2296] LOADER: Overriding Python's sys.path
[2296] LOADER: Post-init sys.path is C:\Users\keem1\AppData\Local\Temp\_MEI93322\base_library.zip;C:\Users\keem1\AppData\Local\Temp\_MEI93322
[2296] LOADER: Setting sys.argv
[2296] LOADER: setting sys._MEIPASS
[2296] LOADER: importing modules from CArchive
[2296] LOADER: extracted struct
[2296] LOADER: callfunction returned...
[2296] LOADER: extracted pyimod01_os_path
[2296] LOADER: callfunction returned...
[2296] LOADER: extracted pyimod02_archive
[2296] LOADER: callfunction returned...
[2296] LOADER: extracted pyimod03_importers
[2296] LOADER: callfunction returned...
[2296] LOADER: Installing PYZ archive with Python modules.
[2296] LOADER: PYZ archive: out00-PYZ.pyz
[2296] LOADER: Running pyiboot01_bootstrap.py
[2296] LOADER: Running pyi_rth_multiprocessing.py
[2296] LOADER: Running sonota.py
Current IPs: ['192.168.43.115']
Select IP address of the WiFi interface:
    0: 192.168.43.115
Select IP address [0]: 0
WiFi SSID: ******
WiFi Password: ******

Using the following configuration:
        Server IP Address: 192.168.43.115
        WiFi SSID: ******
        WiFi Password: ************
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
.......Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "accept": "post",
    "deviceid": "1000c76d44",
    "apikey": "2145c540-380b-4b00-9a9a-e02e863fbb2b",
    "chipid": "00D98FA8"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "ssid": "******",
    "password": "************",
    "serverName": "192.168.43.115",
    "port": 8443,
    "version": 4
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.43.115) is not assigned to any interface on this machine.
** Please change WiFi network to ****** and make sure 192.168.43.115 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
......Current IPs: ['192.168.0.155']
...................................................................................................................................................................................................................................................................................................................................................................................................................................Quitting.
[2296] LOADER: OK.
[2296] LOADER: Cleaning up Python interpreter.
[9332] LOADER: Back to parent (RC: 0)
[9332] LOADER: Doing cleanup
[9332] LOADER: Freeing archive status for C:\Users\keem1\Desktop\SonOTA-master\sonota.exe`

aurelian17, 2020-12-30:
I have the same issue. Are there any updates on this topic? Thanks

scdebaay, 2021-02-13:
I'm having the same issue. Tried with several Sonoff Basic switches. No dice. Have switched all firewall off and the webserver at Stage 2 seems to come up, but the device does not connect (2 short flashes). Any ideas? Would really like to get this to work with my Domoticz implementation.

iminet, 2021-02-13:
I've stopped trying use SonOTA, this is a f*cking bad app, without any support, without any ability to listen to the users.
I flashed Tasmota via serial port in 3 minutes without any issues.
I suggest to all of you, use serial flashing instead. Easy super fast and independent.

pirracas77, 2021-02-13:
> I've stopped trying use SonOTA, this is a f*cking bad app, without any support, without any ability to listen to the users.
> I flashed Tasmota via serial port in 3 minutes without any issues. A few days ago I flashed the 5th Sonoff BasicR2 via serial. No problems of them.
> I suggest to all of you, use serial flashing instead. Easy super fast and independent.
> 
> You need to flash a Sonoff device via serial once. After already using Tasmota, you can flash updates via Tasmota web admin site over the internet.

yes, it is easy if you have an accessible pinout. SonOta was a good idea when this pinout is not accessible .

iminet, 2021-02-13:
> yes, it is easy if you have an accessible pinout. SonOta was a good idea when this pinout is not accessible .

Which version of Sonoff has no pinout? I've never see that. The earlier version had pinout on the middle of the PCB (not so easy to access) while the recent edition has on the edge of the PCB (much easier to use).

pirracas77, 2021-02-13:
> Which version of Sonoff has no pinout? I've never see that. The earlier version had pinout on the middle of the PCB (not so easy to access) while the recent edition has on the edge of the PCB (much [...]

Not all the devices based on ESP8266 are sonoff.

I have this one and it is a good example about what I say.

https://es.aliexpress.com/item/4000490400643.html

[image]

iminet, 2021-02-13:
Yeah, it's true, I've only met Sonoff devices before. I have to correct my advice: use serial flashing only if you have devices with serial pinout 😁

mirko, 2021-02-13:
> I've stopped trying use SonOTA, this is a f*cking bad app, without any support, without any ability to listen to the users.

Dear @iminet,

I reverse-engineered the update process for Sonoff software running devices in in early 2017, documented my findings and wrote a proof of concept program allowing flashing respective devices without UART connection in my spare time.

Figuring it might be useful to others I cleaned it up and published it on GitHub as part of this very repository in May 2017.

I did not expect this project blowing up as it did, yet I tried to put as much effort as somehow possible to deal with reported issues, new versions, new devices, enhancing the code base.

At some point, where I was not able to cope with the volume of users reporting issues anymore, @sillyfrog stepped in and basically took over maintenance (thank you so much at this point!).

Meanwhile, Sonoff fixed the security issue which we exploited in order flash the devices via their update routines and more and more devices were now "immune".

Hence, already in January 2018 I issued a big fat warning that SonOTA does not work anymore current firmware versions (https://github.com/mirko/SonOTA/commit/3869821a7e69b6754f31d834a335ce01e8913cf6).
It is still there, in big fat letters at the very front page of this repository.

**This project is to be considered as deprecated.**

@iminet, this is a project which I did not expect to raise that amount of attention, but - given the feedback we received - apparently quite a number of people found it useful in the past. I'm sorry that it disappoints you - yet, nobody forced you to use it.

We developed this software back then in our spare time, never asked for or received any donations of any kinds, but wanted to share it with those who might find it useful.

All the best

  mirko

iminet, 2021-02-13:
Hey @mirko,

I'm very sorry for the bad words I used, you're right, I'm so sorry.
I know, many of developers make their business only for serving the community and they don't need any awards, just two words 'thank you'.
I'm personally using many of community-maintained stuffs, I swear.

But to be honest, it was a little annyoing while the internet is full of your stuff and everyone recommended it, not only in the past. Several forum posts, blogs etc said, after a little pause, SonOTA works again.
Yes, I saw your warning and I thoungt, you forgot to remove it.
After my bug report I waited a lot of weeks without any official response, and finally I figured it out, I'll use serial instead. It worked as simple as I can't expect and left to wait to any news about SonOTA.

But I understand, many of users had a huge help of it. I hope, your project (and its ability) will reborn on a day.